Squiddrone/Squiddrone #149: Model Com Component

elheck, 2020-01-18:


c0nwild, 2020-04-14:
Revision to review: 34, Apr 14, 2020 22:18:40

c0nwild, 2020-04-15:
Revision for review: 35, 2020 Apr. 15, 21:16:12

elheck, 2020-04-18:
In Com Class Diagram:
ComMessageBuffer private attribute queue should be a std::array if the size is known at compile time [see this](https://github.com/elheck/Squiddrone/wiki/Guideline-for-embedded-modern-CPP), because a std::array is allocated on the stack. Dependend methods might need to be changed as well.

Enumerations should have public attributes instead of enum literals, because in cpp we use enum classes which are closer to classes then C enums.

put_data_packet in NRF24L01 overloads put data packet of ComInterface which is basic Inheritance and not an implementation of an interface (think basic inheritance vs abstract base class. Abstract base class being an interface)

An implementation of an interface should be usable by only calling methods that are also defined in the interface and not by additional public methods defined in the implementation of the interface. See [InertialMeasurementInterface and InertialMeasurement](https://github.com/elheck/Squiddrone/wiki/UML-class-IMU) or [ThrustControllerInterface and ThrustController](https://github.com/elheck/Squiddrone/wiki/UML-class-propulsion).
Otherwise another implementation of the interface would mean major changes in the application layer.

SEQ handle irq: LGTM

Address cofiguration phase state diagram: LGTM

get com data activity diagram: LGTM

Did I miss a diagram?

Thank you for modelling such a complex component!

c0nwild, 2020-04-18:
Pleas have a look at spi transaction activity diagram (referenced from call behaviour actions of type 'send command'.

elheck, 2020-04-18:
spi transaction activity diagram: LGTM

elheck, 2020-04-22:
Class diagramm at 20:08 o'clock (22.04.2020): 
- What kind of type is  queue<vector<uint8_t>>  from ComMessageBuffer?
- Has vector<uint8_t> which is used for data and payload a size that is known at compile time? If yes, please use array<uint8_t>, because it is stack allocated.

elheck, 2020-04-22:
LGTM Ready for implementation

closed 2020-04-22 by elheck