Squiddrone/Squiddrone #150: Model IMU Component

elheck, 2020-01-18:


ThorsHamster, 2020-04-11:
Updated like suggested. Please take a look, if its better now.
Possible improvements: 
Removal of GenericIMU and use only of MPU9250?!
Add generic Sensorclass and use Gyroscope, Accelerometer etc als childs of this class.

elheck, 2020-04-11:
I have some points to discuss about:

1. In InertialMeasrementUnit the private member is called generic_i... and is of type GenericInertialMeasurementUnit. Shouldn't it be a concrete IMU witch will be identified by the GenericInertialMeasurementUnit abstract interface? What I am suggesting is to rename it from generic_inertialmeasurementunit to inertial_measurement_unit in order not to confuse the concrete object with its abstract interface.

2. The class InterialMeasurementUnit is the accessor for all concrete IMUs defined via their interface GenericInterialMeasurementUnit. I think InterialMeasurementUnit should be named InertialMeasurement and it's interface InertialMeasurementInterface, because this is what this class does. It provides a facade for InterialMeasurements via something that is hidden. I am willing to discuss this, but this is my oppinion.

3. At what point do you know the actual IMU, MPU9250 in this case? Who creates an object of if?

4. I agree on the generic Sensorclass now. It narrows down the interface between MPU and Sensors and will also save some effort in implementation.

5. good job, the overall structure is very much to my liking

ThorsHamster, 2020-04-11:
[image]
@elheck 
What do you think about this approach?

elheck, 2020-04-11:
Yes this approach is good.

closed 2020-04-12 by ThorsHamster