Squiddrone/Squiddrone #193: Wiki entry - unified doxygen documentation style

elheck, 2020-04-30:


c0nwild, 2020-05-03:
First draw here: https://github.com/elheck/Squiddrone/wiki/Doxygen-Style-Guide
pls. review

elheck, 2020-05-04:
How about single line comments above the line of code? By doing this it becomes easier to read, because it is more compact and left side aligned.

Doxygen Documentation Generator generates the comments that all fist * -signs are aligned.
See [motor.hpp](https://github.com/elheck/Squiddrone/blob/dev/squidDrone/src/propulsion/motor.hpp) for reference.

In your draft, the * signs are aligned with the opening /

c0nwild, 2020-05-04:
Alignment of multiline comments in examples is now correct.
Good idea, placing the single line comments above the code. Done.

elheck, 2020-05-05:
LGTM

closed 2020-05-05 by elheck